balena-io-library/base-images #660: balenalib/raspberrypi4-64-golang:1.15 images missing

rickardgranberg, 2020-09-24:
Can't find any Go 1.15 images on Docker hub.

I can find Dockerfile(s) here for all distros with Go 1.15 in them, for example: [balena-base-images/golang/raspberrypi4-64/debian/buster/1.15/build/Dockerfile](balena-base-images/golang/raspberrypi4-64/debian/buster/1.15/build/Dockerfile)

Am I missing something?

rickardgranberg, 2020-10-09:
It's been 15 days, any information about this problem?

nghiant2710, 2020-10-09:
@rickardgranberg, so sorry for the inconvenience. Due to some issues with our build servers so the image you're looking for is missing. However, as we're moving the latest tag for Golang base images to `v1.15.2` so the `balenalib/raspberrypi4-64-golang:1.15.2` is now available.

Can you please give it a try?

rickardgranberg, 2020-10-09:
It works now. Thank you very much!

closed 2020-10-09 by rickardgranberg