balena-io-library/base-images #661: python 2.7.18 manifest missing??

ealmonte32, 2020-10-06:
What am I doing wrong..? following the exact instructions on this page:
https://www.balena.io/docs/reference/base-images/base-images/

This is the first line of the dockerfile:

FROM balenalib/%%BALENA_MACHINE_NAME%%-debian-python:2.7.18-buster-run AS build

This is the error:
manifest for balenalib/raspberrypi3-debian-python:2.7.18-buster-run not found: manifest unknown: manifest unknown

tried some variations of the debian-python:2.7.18-buster part and still no good...

ealmonte32, 2020-10-06:
I will answer myself.. it seems like the instructions need a little bit of update because if you tell users to put the specific lang version but then dont sort of symlink 2.7.18 to equal "latest" which is not the actual latest, then this confuses the user building the dockerfile..i cannot be the only one that gets confused by the word "latest" to mean 2.7's latest and not 3.x as latest..

ealmonte32, 2020-10-06:
actually... I thought latest would go and run latest "2.7.x" but it actually does build a 3.8 version of python which is NOT what I need...

nghiant2710, 2020-10-07:
@ealmonte32, thank you for the report, I will check our build servers to see if there is something wrong with building this version and get back to you as soon as possible. 

By the way, the latest tag for Python base images is now pointing to `v3.8.x` as Python 2.x is now deprecated.

ealmonte32, 2020-10-07:
@nghiant2710 
Thanks for looking into it, and yes I understand python 2.x is deprecated, but some projects still use it thus we need it.

Also, this is why I  think the documentation on the balena article explaining how to properly use the FROM field in the Dockerfile needs to be updated since it now incorrectly points out that the "latest" tag will point to 2.7.17 even while showing v3.x as well on the same page, and users who start creating the Dockerfile will try the incorrect image label and waste time.

I understand it is a pain to keep track of all the different version tags but maybe just write somewhere on that page that the "latest" tag will always point to the latest available version of the language pack being called, and that if users want to use any other available version that are actually hosted you can point them to the github repo location where they can search for the different builds thus they will know for sure which specific version is available.

Thanks again.

[image]

[image]

[image]