balena-io-library/base-images #669: dotnet:3.1-sdk manifest missing for armv7hf/raspberrypi3

flec, 2020-11-27:
Dockerfiles with
`FROM balenalib/%%BALENA_MACHINE_NAME%%-dotnet:3.1-sdk-build` 
or 
`FROM balenalib/%%BALENA_ARCH%%-dotnet:3.1-sdk-build` 
produce the flowing errors with raspberrypi3 respectively.

```
manifest for balenalib/raspberrypi3-dotnet:3.1-sdk-build not found: manifest unknown: manifest unknown

manifest for balenalib/armv7hf-dotnet:3.1-sdk-build not found: manifest unknown: manifest unknown
```
raspberrypi4-64/aarch64 works with the same FROM clause. Are those manifests missing?
According to the [documentation](https://www.balena.io/docs/reference/base-images/base-images/#supported-architectures-distros-and-languages) those architectures should be supported.

`FROM balenalib/%%BALENA_MACHINE_NAME%%-dotnet:3.1-run` works for both architectures/raspberrypies

nghiant2710, 2020-11-27:
@flec, thank you for the report, we will check our base images and get back to you as soon as possible.

nghiant2710, 2020-12-09:
hey @flec, due to some QEMU limitation with .NET Core, `3.1-sdk` can't be installed properly on armv7 Debian Buster (both `balenalib/raspberrypi3-dotnet:3.1-sdk-build` and `balenalib/armv7hf-dotnet:3.1-sdk-build` are using Debian Buster as the OS) So until we get this sorted, can you please use the `balenalib/raspberrypi3-dotnet:3.1-sdk-stretch-build` or `balenalib/armv7hf-dotnet:3.1-sdk-stretch-build` (both built off Debian Stretch) instead?

Please let us know if you have any problem with the image so we can help.

flec, 2020-12-18:
Thanks @nghiant2710, building in `*-dotnet:3.1-sdk-stretch-build` does work.
With the following setup we are able to run it in `*-dotnet:3.1-buster-run`:
```
# build
FROM balenalib/%%BALENA_ARCH%%-dotnet:3.1-sdk-stretch-build as build

...

# run
FROM balenalib/%%BALENA_ARCH%%-dotnet:3.1-buster-run

...
```

closed 2021-01-13 by nghiant2710