balena-io-library/base-images #672: raspberrypi3 image for linux/amd64?

PLCHome, 2020-12-11:
Why is this warning displayed, the raspberry PI3 has never had an AMD processor?

I start the image on a PI4 with 32 bit raspian buster inatalled.
```
pi@HomeBerry2:/docker/basebuster $ sudo docker run -d --name test balenalib/raspberrypi3:buster
WARNING: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm/v7) and no specific platform was requested
4c0006b1f8b04a3e0f538d7a75ef70154b1cec54c681c8eda4aab174b078e187
pi@HomeBerry2:/docker/basebuster $ sudo docker rm -f test

```

The container is for arm v7 and not amd64:
```
Raspbian GNU/Linux 10 (buster)
Linux busterimage 5.4.79-v7l+ #1373 SMP Mon Nov 23 13:27:40 GMT 2020 armv7l GNU/Linux

```

RichardFoo, 2021-03-29:
It would be nice to see a comment on this after 3 months, even though it seems to be just a cosmetic warning.  Reading Google results for other platforms (e.g., Portainer), it appears to be caused by an oversight in the image building.  See what Docker Inspect reports:

**docker image inspect balenalib/rpi-alpine-python:latest-build**
        "Architecture": "amd64",
        "Os": "linux",

PLCHome, 2021-03-29:
I still have the same message as I did 3 months ago. I don't think it's nice to close the ticket without fixing the bug.

```
$ sudo docker pull balenalib/raspberrypi3:buster
buster: Pulling from balenalib/raspberrypi3
db574d82360c: Already exists 
5c053f1a5563: Pull complete 
74c1ec065164: Pull complete 
62813535aed2: Pull complete 
a61b6d45fb8f: Pull complete 
a1952b570da4: Pull complete 
6797b421b336: Pull complete 
1c4c21365cea: Pull complete 
5e24a751064b: Pull complete 
034f0332b32f: Pull complete 
2497d7d19338: Pull complete 
7b2f55b16bdf: Pull complete 
537329b7b58a: Pull complete 
a01c25f432e7: Pull complete 
c7502c2a64f4: Pull complete 
86b4b5424518: Pull complete 
9b00fe0e4470: Pull complete 
Digest: sha256:95161b9b8a5dc58fd4c55838d6a616ed3c95df5f6aa5117df03aa7cdbee44af6
Status: Downloaded newer image for balenalib/raspberrypi3:buster
docker.io/balenalib/raspberrypi3:buster

$ sudo docker run -d --name test balenalib/raspberrypi3:buster
WARNING: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm/v7) and no specific platform was requested
87c85481dea5be505819b913f1a61b02038f3c89a56e1c0192da1d90d536c7b9
```

webash, 2021-04-06:
Also having this problem trying to build balena-flic, a project based on the balenalib/armv7hf-debian (renemarc/balena-flic#3) on the Raspberry Pi 4, running Raspberry Pi OS (32bit) and Docker 20.10.04. Balena images all report their architecture as linux/amd64.

nghiant2710, 2021-04-06:
@PLCHome, @RichardFoo, @webash sorry for the late response, we're still working on this. Basically, our base images builders are running old Docker version which doesn't allow setting the architecture in the manifest, we're working on moving to the new ones which will correctly set the architecture.

I will let you know when it's done, thank you for the patience.

webash, 2021-04-07:
Thanks for the response. Let me know if anything goes up for testing - happy to do so.