balena-io-library/base-images #687: balenalib/raspberrypi4-64-golang:1.16 images missing

rickardgranberg, 2021-03-26:
Can't find any Go 1.16 images on Docker hub.

I can find Dockerfile(s) here for all distros with Go 1.16 in them, for example: https://github.com/balena-io-library/base-images/blob/master/balena-base-images/golang/raspberrypi4-64/debian/buster/1.16/build/Dockerfile

This same problem happened for 1.15 images iirc.

rblenkinsopp, 2021-08-06:
They are also missing for the `raspberry-pi` (v1 & zero) architectures.

paulburlumi, 2021-08-23:
Fixed?
https://github.com/balena-io-library/base-images/blob/master/balena-base-images/golang/raspberrypi4-64/debian/buster/1.16.7/build/Dockerfile

rickardgranberg, 2021-08-24:
> Fixed?
> https://github.com/balena-io-library/base-images/blob/master/balena-base-images/golang/raspberrypi4-64/debian/buster/1.16.7/build/Dockerfile

No, the problem is that the images does not exist on Docker Hub. For some reason they aren't pushed...

nghiant2710, 2021-08-24:
Hi all, thank you for the report, we acknowledged the issue and will add the missing base-images soon.
I will let you know when they are available.

nghiant2710, 2021-08-30:
Just want to let you know that the Go v1.16 and v1.17 base images are now available, please let us know if you need further support.

closed 2021-08-30 by nghiant2710