balena-io-library/base-images #688: aclocal fails to run during crossbuilds

roflcoopter, 2021-03-26:
I am having issues building several libraries from source which are using ```autogen.sh``` such as [libzmq](https://github.com/zeromq/libzmq) for a raspberrypi3 on my amd64 machine.

I end up getting this error on build:
```
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal -I config --force -I config
aclocal: error: echo failed with exit status: 1
autoreconf: aclocal failed with exit status: 1
autogen.sh: error: autoreconf exited with status 0
```
Dockerfile:
```Dockerfile
FROM balenalib/raspberrypi3-debian:buster-build as build
RUN [ "cross-build-start" ]
WORKDIR /tmp/workdir

ENV LIBZMQ_VERSION=4.3.2 \
  PREFIX=/opt/ffmpeg

RUN   apt-get -yqq update && \
  apt-get install -yq --no-install-recommends cmake libtool

RUN \
  DIR=/tmp/libzmq && \
  mkdir -p ${DIR} && \
  cd ${DIR} && \
  curl -sLO https://github.com/zeromq/libzmq/archive/v${LIBZMQ_VERSION}.tar.gz && \
  tar -xz --strip-components=1 -f v${LIBZMQ_VERSION}.tar.gz && \
  ./autogen.sh && \
  ./configure --prefix="${PREFIX}" && \
  make -j $(nproc) && \
  # make check && \
  make -j $(nproc) install && \
  rm -rf ${DIR}

RUN [ "cross-build-end" ]

```

I messed around a little bit, and this Dockerfile builds just fine:
```Dockerfile
FROM balenalib/raspberrypi3-debian:buster-build as build
RUN [ "cross-build-start" ]
WORKDIR /tmp/workdir

ENV LIBZMQ_VERSION=4.3.2 \
  PREFIX=/opt/ffmpeg

RUN   apt-get -yqq update && \
  apt-get install -yq --no-install-recommends cmake libtool

RUN [ "cross-build-end" ]

RUN \
  DIR=/tmp/libzmq && \
  mkdir -p ${DIR} && \
  cd ${DIR} && \
  curl -sLO https://github.com/zeromq/libzmq/archive/v${LIBZMQ_VERSION}.tar.gz && \
  tar -xz --strip-components=1 -f v${LIBZMQ_VERSION}.tar.gz && \
  ./autogen.sh

RUN [ "cross-build-start" ]

RUN \
  DIR=/tmp/libzmq && \
  mkdir -p ${DIR} && \
  cd ${DIR} && \
  ./configure --prefix="${PREFIX}" && \
  make -j $(nproc) && \
  # make check && \
  make -j $(nproc) install && \
  rm -rf ${DIR}

RUN [ "cross-build-end" ]
```

Any suggestions? Can i solve this somehow?

nghiant2710, 2021-04-02:
hey @roflcoopter, we use QEMU to handle emulation when cross-building so it could be the problem with `aclocal` command and QEMU don't play well with others. We're using latest QEMU with some in-house patches so I think you should stay with the current workaround until we have newer QEMU released.

roflcoopter, 2021-04-06:
> hey @roflcoopter, we use QEMU to handle emulation when cross-building so it could be the problem with `aclocal` command and QEMU don't play well with others. We're using latest QEMU with some in-house patches so I think you should stay with the current workaround until we have newer QEMU released.

Okay i see, thanks for your reply!