balena-io-library/base-images #695: manifest for balenalib/raspberrypi4-64-debian-node:14.16.1-buster-build not found in local mode

cailloumajor, 2021-05-06:
Despite the existence of:
 - https://github.com/balena-io-library/base-images/blob/92acdbdb6c3bb002036da7ef4759c339d93e73ba/balena-base-images/node/raspberrypi4-64/debian/buster/14.16.1/build/Dockerfile
 - https://github.com/balena-io-library/official-images/blob/0e7cdfe8099bad9cf7dfca6f162e946fc6ea6504/library/raspberrypi4-64-debian-node#L19

After reading the docs about base images naming scheme and exploring the base images repository, I don't understand how users can find which versions are usable. From my point of view, I feel lost...

cailloumajor, 2021-05-08:
Just tried after disabling local mode, same problem:
[image]

If I set the image tag to `14`, I get an image with [Node.js v14.7.0](https://github.com/nodejs/node/blob/master/doc/changelogs/CHANGELOG_V14.md#2020-07-29-version-1470-current-mylesborins-prepared-by-ruyadorno), quite out-of-date !

maggie0002, 2021-05-22:
I just came across the same issue. There appears to be some gaps in the available images for node (perhaps others too, but haven't explored others). The effect of course is that I can't use %%BALENA_MACHINE_NAME%% in my Dockerfiles as the file is only available for some architectures. 

Available:

```
balenalib/aarch64-alpine-node:14.16-3.13-build
balenalib/armv7hf-alpine-node:14.16-3.13-build
```

Not available:

```
balenalib/armv6hf-alpine-node:14.16-3.13-build
balenalib/amd64-alpine-node:14.16-3.13-build
```

closed 2022-06-22 by nghiant2710