balena-io-library/base-images #696: Missing Base Images for some Architectures

dtischler, 2021-05-22:
@nghiant2710 Can you help shed some light on why some Base images don't exist for all architectures?  Here is an example: 

Available:

```balenalib/aarch64-alpine-node:14.16-3.13-build
balenalib/armv7hf-alpine-node:14.16-3.13-build
```

Not available:

```
balenalib/armv6hf-alpine-node:14.16-3.13-build
balenalib/amd64-alpine-node:14.16-3.13-build
```

And, could we please have those `alpine-node:14.16-3.13-build` created for armv6 and amd64, please?   :-)

dtischler, 2021-05-22:
cc:  @ maggie0002

maggie0002, 2021-06-03:
Another one.

Available:

balenalib/amd64-alpine-python:3.8.10-3.13

Not available:

balenalib/raspberrypi4-64-alpine-python:3.8.10-3.13

maggie0002, 2021-06-07:
Linked issue on Balena Forums for reference: https://forums.balena.io/t/missing-images-from-balenalib/319708/8

wrobell, 2021-06-11:
Similar to #686.

maggie0002, 2021-06-14:
> Similar to #686.

And I think other similar posts:

https://github.com/balena-io-library/base-images/issues/695
https://github.com/balena-io-library/base-images/issues/691
https://github.com/balena-io-library/base-images/issues/687
https://github.com/balena-io-library/base-images/issues/680
https://github.com/balena-io-library/base-images/issues/661
https://github.com/balena-io-library/base-images/issues/606
https://github.com/balena-io-library/base-images/issues/602

nghiant2710, 2021-06-15:
Hey @maggie0002, just want to let know that all the images you mentioned are now available. So sorry for the inconvenience, I will work on other issues as well and keep this issue open so everyone can report if there is a missing tag and we will try to resolve as soon as possible.

zartdinov, 2021-10-04:
@nghiant2710 Hi! I can't find tag for this Dockerfile:
https://github.com/balena-io-library/base-images/blob/master/balena-base-images/python/raspberrypi4-64/debian/bullseye/3.9.7/build/Dockerfile

closed 2022-06-22 by nghiant2710