balena-io-library/base-images #700: QEMU and CMake Version 3.16.3 (raspberry-pi:buster/raspberry-pi2:buster)

Paulchen-Panther, 2021-06-16:
When using the images raspberry-pi:buster or raspberry-pi2:buster to cross compile a CMake project, CMake does not find the C++/C compiler. The error message is the following:

```
-- CMake Version: 3.16.3
CMake Error at /usr/share/cmake-3.16/Modules/CMakeCompilerIdDetection.cmake:26 (list):
  list sub-command REMOVE_ITEM requires two or more arguments.
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/CMakeDetermineCompilerId.cmake:211 (compiler_id_detection)
  /usr/share/cmake-3.16/Modules/CMakeDetermineCompilerId.cmake:230 (CMAKE_DETERMINE_COMPILER_ID_WRITE)
  /usr/share/cmake-3.16/Modules/CMakeDetermineCompilerId.cmake:32 (CMAKE_DETERMINE_COMPILER_ID_BUILD)
  /usr/share/cmake-3.16/Modules/CMakeDetermineCCompiler.cmake:116 (CMAKE_DETERMINE_COMPILER_ID)
  CMakeLists.txt:5 (PROJECT)
 
-- The CXX compiler identification is unknown
```

Apparently, Balenalib uses the backports repositories under Buster. When the CMake version is reset to Buster (3.13.4), CMake also finds the C++/C compilers. Since I don't know how far Balenalib updates the QEMU binaries, I can only limit the error to CMake.

Would it be possible to fix this bug in the images without having to downgrade the CMake version?

Thanks in advance for your help.

tomikais, 2021-08-21:
Hi @Paulchen-Panther,

I have the same issue when [building](https://gitlab.com/tomikais/fotobox/-/jobs/1520824234) my project.
Do you have a workaround for it? If yes, can you please share it :-)

Thanks, Thomas

Paulchen-Panther, 2021-08-21:
Hello @tomikais 

As I have seen you have already downgraded the cmake/cmake-data version to 3.13.4-1. In my case [(hyperion.docker.ci)](https://github.com/hyperion-project/hyperion.docker-ci/commit/dec107c9cfd793163f0baef16ec1a8ed9196ab3b) this helped.

Greetings, Markus

tomikais, 2021-08-21:
Thanks for your fast reply and for your fix. I choose to [build it with 'qmake'](https://gitlab.com/tomikais/fotobox/-/commit/a3d3178da81f6508ab9bb11ba5ad9fd1a1e77db2) instead of CMake.