balena-io-library/base-images #701: Java 16 images

AlexProgrammerDE, 2021-06-19:
Hi there! I would like to suggest Java 16 images to be an option to choose. I need it in my project [balena-minecraft-server](https://github.com/AlexProgrammerDE/balena-minecraft-server) as Java 16 is now an official requirement to get the server software working on 1.17. Should hopefully not be too hard as Java 8 and 11 images already exist. Thanks!

nghiant2710, 2021-07-05:
hey @AlexProgrammerDE just want to let you know that the openjdk 16 (JDK) base images for amd64 and aarch64 devices are available.

closed 2021-07-05 by nghiant2710

AlexProgrammerDE, 2021-07-05:
Hi @nghiant2710 could these images also get published for the specific device types? That is what my fleet is relying on at the moment.

nghiant2710, 2021-07-05:
@AlexProgrammerDE what are you target device types? `amd64 and aarch64 devices` mean all specific devices with amd64/aarch64 architecture will have openJDK 16 base images

AlexProgrammerDE, 2021-07-05:
@nghiant2710 according to dockerhub the last update for raspberrypi4-64 was 5 months ago: https://hub.docker.com/r/balenalib/raspberrypi4-64-debian-openjdk/tags?page=1&ordering=last_updated

nghiant2710, 2021-07-05:
@AlexProgrammerDE, I will check our build servers to see if anything is wrong but the PR was merged last week so it should be available now.

nghiant2710, 2021-07-05:
@AlexProgrammerDE, I restarted the builds, the openJDK 16 images should be there, Can you give it a try?

AlexProgrammerDE, 2021-07-05:
@nghiant2710 I just pushed successfully to my fleet. Seems to work fine. Thanks!