balena-io-library/base-images #702: Java 16

taedium8vitae, 2021-06-19:
Please update the java version to 16.

matthewvcarey1, 2021-06-26:
I can see the message in the log:
Checking server JAR... Found a valid server file. It's called: paper.jar. Use ENABLE_UPDATE to update.
26.06.21 09:00:19 (+0100)  mc-server  Starting JAR file with: 1G of RAM
26.06.21 09:00:19 (+0100)  mc-server  Minecraft 1.17 requires running the server with Java 16 or above. Download Java 16 (or above) from https://adoptopenjdk.net/
26.06.21 09:00:30 (+0100) Service exited 'mc-server sha256:b6b98a63ed7c02218ebc8334012b42766ded380eb5cce85e14974e1650a17431'
26.06.21 09:00:30 (+0100) Restarting service 'mc-server sha256:b6b98a63ed7c02218ebc8334012b42766ded380eb5cce85e14974e1650a17431'
26.06.21 09:00:31 (+0100)  mc-server  Setting device hostname to: balenaminecraftserver

Either make it easy to just install the correct java or fix it really quickly

I am going to give up and just flash with Ubuntu arm64 Server and install minecraft server myself, this is harder than doing it by hand.

AhsenBaig, 2021-06-30:
Duplicate to: #701

nghiant2710, 2021-07-05:
Hey @taedium8vitae @matthewvcarey1 @AhsenBaig, just want to let you know that the openjdk 16 (JDK) base images for amd64 and aarch64 devices are available.

closed 2021-07-05 by nghiant2710

AhsenBaig, 2021-07-05:
Rebooted device on every change. Manually updated paper.jar as ENABLE_UPDATE wasn't working properly. 

Set the Device Variable:
ENABLE_UPDATE `TRUE`

`04.07.21 23:06:00 (-0700)  mc-server  Forcing server update
04.07.21 23:06:00 (-0700)  mc-server  Downloading 1.17 build 75...
04.07.21 23:06:04 (-0700)  mc-server  Starting JAR file with: 8G of RAM
04.07.21 23:06:05 (-0700)  mc-server  Minecraft 1.17 requires running the server with Java 16 or above. Download Java 16 (or above) from https://adoptopenjdk.net/`

Even updated:
belenaOS and supervisor

nghiant2710, 2021-07-05:
@AhsenBaig what docker base image are you using?

AhsenBaig, 2021-07-05:
I am using:
`FROM balenalib/%%BALENA_MACHINE_NAME%%-debian-openjdk:latest-buster`

If you want to see the [full project build](https://github.com/AhsenBaig-forks/balena-minecraft-server).

Java application [mc-server](https://github.com/AhsenBaig-forks/balena-minecraft-server/tree/master/mc-server)

Let me know if you want any testing/configuration changes.