balena-io-library/base-images #731: Date inside a container is 01/01/1970

limes007, 2021-11-09:
I pulled a new image of balenalib/raspberrypi3-python and got the following problem

```
balenalib/raspberrypi3-python                       latest    408c83c4b098   18 hours ago     271MB
balenalib/raspberrypi3-python                       <none>    301374a87406   7 weeks ago      258MB

$ docker run --rm -it 301374a87406 date
Tue Nov  9 15:20:56 UTC 2021
$ docker run --rm -it 408c83c4b098 date
Thu Jan  1 00:00:00 UTC 1970
```

limes007, 2021-11-10:
Latest-Tag seems to provide bullseye instead of buster, maybe this is the problem, as my host is running buster. This is strange, as https://www.balena.io/docs/reference/base-images/base-images-ref/ says that buster is latest.

```
$ docker run --rm -it balenalib/raspberrypi3-python date
Thu Jan  1 00:00:00 UTC 1970
$ docker run --rm -it balenalib/raspberrypi3-python:latest date
Thu Jan  1 00:00:00 UTC 1970
$ docker run --rm -it balenalib/raspberrypi3-python:latest-buster date
Wed Nov 10 18:02:49 UTC 2021
```

nghiant2710, 2021-11-12:
@limes007 thank you for the report, the balenalib latest Debian image points to `bullseye` (same as official Debian), the document is a bit outdated and we will fix it very soon.

Regarding the date inside container, as far as I know docker containers don't maintain a separate clock, it's identical to the Linux host since time is not a namespaced value and we don't have anything that change/update the time inside our Dockerfile hence it should something on the Docker engine or your OS, can you let us know more details about your environment so I can try to reproduce it?

The `date` command output is correct on my end as below
```
nghiant2710@misc1:~$ docker run -ti --rm balenalib/raspberrypi3-python:latest bash
root@c985c6a7d469:/# date
Fri Nov 12 04:41:46 UTC 2021
root@c985c6a7d469:/# cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

limes007, 2021-11-12:
Hi @nghiant2710 thank you for your reply.

My assumption was that the date is correct if the host and container are using the same master distribution version. I am using buster on the host and there is no data problem with the buster image. You use bullseye on host and container and have no problem either. 

My host runs Raspiban buster
```
$ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

Docker version is 20.10.10
```
$ docker --version
Docker version 20.10.10, build b485636
```

What else do you want to now about my environment?