balena-io-library/base-images #735: Running shell script as CMD fails on alpine image.

BlackDex, 2021-11-19:
Hello there,

I had some issues using the `balenalib/armv7hf-alpine:3.14` image and simply let it run `/start.sh` using that as the `CMD` option.
This does work on a Debian based image btw.

When i checked the `/usr/bin/entry.sh` file it executes the following: `if CMD=$(command -v "$1" 2>/dev/null); then`.
If i execute `command -v "/start.sh"` from an `sh` shell, it get a correct response.
But, if i execute that same command from a `bash` shell, it doesn't return anything.

If i change the shebang for the `entry.sh` from `/bin/bash` to `/bin/sh` it works again.

Is there anything you can do to get this working out-off-the-box?

Thanks in advance.

Small example (application logic removed here):
**Dockerfile:**
```dockerfile
FROM balenalib/armv7hf-alpine:3.14

# hadolint ignore=DL3059
RUN [ "cross-build-start" ]

# Create data folder and Install needed libraries
RUN mkdir /data \
    && apk add --no-cache \
        openssl \
        tzdata \
        curl \
        ca-certificates

# hadolint ignore=DL3059
RUN [ "cross-build-end" ]

VOLUME /data
EXPOSE 80

WORKDIR /

COPY start.sh /start.sh

# Configures the startup!
CMD ["/start.sh"]
```
**start.sh:**
```bash
#!/bin/sh

echo "Hello World!"
```

BlackDex, 2021-11-19:
Small addition, it also does not return anything if you run `command -v /bin/sh` using the `bash` shell.
While it does using the `sh` shell.

klutchell, 2021-12-21:
This is due to the wrapper shim we have around `sh` that is executed on first run. If you bypass `sh` by running `bash` directly, the binary `/bin/sh` is still our wrapper, and the real executable is at `/bin/sh.real`.

Essentially, until it has been executed once `/bin/sh` is not a true shell but is a wrapper that doesn't behave exactly like a shell. Once it has been executed then the real shell binary is restored.

klutchell, 2022-07-15:
Should be resolved by https://github.com/balena-io/contracts/pull/310

BlackDex, 2023-03-31:
It is. And working for a while now. Thanks.
I'll close this issue.

closed 2023-03-31 by BlackDex