balena-io-library/base-images #741: balenalib/rpi-raspbian fails to build on apple silicon Macs.

cosnicolaou, 2021-12-21:
The following Dockerfile fails with QEMU illegal instruction error when run on an apple silicon Mac.

```
FROM balenalib/rpi-raspbian:bullseye-20210925
USER root
RUN apt-get update -y && apt-get upgrade -y
```

builder555, 2021-12-24:
Adding ENV QEMU_CPU=max in the dockerfile fixes the issue:
```
FROM balenalib/rpi-raspbian:bullseye-20210925
ENV QEMU_CPU=max
USER root
RUN apt-get update -y && apt-get upgrade -y
```

cosnicolaou, 2021-12-28:
thanks, that does indeed work. Can you explain why? Thanks again!

builder555, 2021-12-28:
[This issue](https://github.com/docker/for-mac/issues/5561) mentions that by qemu defaults to "core2duo" cpu, when nothing is specified. I personally, don't understand a lot of it, but what stood out is that enabling all features (i.e. "max") should work every time.

closed 2021-12-29 by builder555