balena-io-library/base-images #742: manifest for balenalib/iot-gate-imx8-node:latest not found: manifest unknown: manifest unknown

npatil-boosterfuels, 2022-01-04:
Not able to get base image for CompuLab iot-gate-imx8 platform during local balena push. So far I have tried iot-gate-imx8-node:latest , iot-gate-imx8-node:14.18 and iot-gate-imx8-debian-node:14.18 with no luck.

Related balena forum topics,
https://forums.balena.io/t/manifest-for-balenalib-iot-gate-imx8-node-latest-not-found-manifest-unknown-manifest-unknown/350806

https://forums.balena.io/t/balenalib-iot-gate-imx8-node-12-18-not-found-manifest-unknown-manifest-unknown/350730
--------------------
manifest for balenalib/iot-gate-imx8-node:latest not found: manifest unknown: manifest unknown

Error: Some services failed to build:
manifest for balenalib/iot-gate-imx8-node:latest not found: manifest unknown: manifest unknown

-----------------------
manifest for balenalib/iot-gate-imx8-node:14.18 not found: manifest unknown: manifest unknown

Error: Some services failed to build:
manifest for balenalib/iot-gate-imx8-node:14.18 not found: manifest unknown: manifest unknown

------------------------
manifest for balenalib/iot-gate-imx8-debian-node:14.18 not found: manifest unknown: manifest unknown

Error: Some services failed to build:
manifest for balenalib/iot-gate-imx8-debian-node:14.18 not found: manifest unknown: manifest unknown

nghiant2710, 2022-01-04:
@npatil-boosterfuels thanks for the report, I will check the base images build servers and get back to you when the images are available.

nghiant2710, 2022-01-05:
Just want to let you know that the `balenalib/iot-gate-imx8-node` base images are available now, please let us know if you find any missing tags.

closed 2022-01-05 by nghiant2710