balena-io-library/base-images #743: 32-bit arm container on aarch64 platform

npatil-boosterfuels, 2022-01-06:
Hello, 

Due to some external node package dependency we are require to use 32-bit arm container on aarch64 platform (iot-gate-imx8). I tried using 'balenalib/armv7hf-node-run' with no luck. is it possible to run 32-bit Debian container on aarch64 platform?

```
[Debug]   Sending request to http://192.168.2.2:48484/v2/local/device-info
[Debug]   Found build tasks:
[Debug]       optimus: build [./optimus-prime]
[Debug]   Resolving services with [iot-gate-imx8|aarch64]
[Debug]   Found project types:
[Debug]       optimus: Standard Dockerfile
[Debug]   Probing remote daemon for cache images
[Debug]   Using 59 on-device images for cache...
[Debug]   Starting builds...
[Debug]   optimus: Image manifest data unavailable for balenalib/armv7hf-node-run
[Debug]   optimus: Image manifest data unavailable for balenalib/armv7hf-node-run
[Build]   [optimus] Step 1/18 : FROM balenalib/armv7hf-node-run as optimus-deps

[Build]   [optimus] Step 1/18 : FROM balenalib/armv7hf-node-run as optimus-deps
Some services failed to build:
	optimus: pull access denied for balenalib/armv7hf-node-run, repository does not exist or may require 'docker login': denied: requested access to the resource is denied

Error: Some services failed to build:
	optimus: pull access denied for balenalib/armv7hf-node-run, repository does not exist or may require 'docker login': denied: requested access to the resource is denied

    at Object.handleError (/usr/local/lib/balena-cli/build/errors.js:163:17)
    at Object.run (/usr/local/lib/balena-cli/build/app.js:105:73)
    at async run (/usr/local/lib/balena-cli/bin/run:20:2)

For further help or support, visit:
https://www.balena.io/docs/reference/balena-cli/#support-faq-and-troubleshooting

```

nghiant2710, 2022-01-06:
@npatil-boosterfuels, looks like you are using wrong base images, it should be `balenalib/armv7hf-node`. So something like `balenalib/armv7hf-node:latest` or `balenalib/armv7hf-node:14.18` should work.

npatil-boosterfuels, 2022-01-06:
@nghiant2710 Thank you for the info. It worked!

closed 2022-01-06 by npatil-boosterfuels