balena-io-library/base-images #746: apt hangs on raspberry images

DrSchottky, 2022-01-19:
I'm trying to run debian images for raspberry pi ( I tried different debian versions for differnt Pi models) but if I start the container

` docker run --rm -it --entrypoint bash  balenalib/raspberry-pi`

and run `apt update` it hangs forever here
```
Get:1 http://archive.raspberrypi.org/debian bullseye InRelease [23.5 kB]
Get:2 http://archive.raspbian.org/raspbian bullseye InRelease [15.0 kB]
0% [Working]
```
I tried it on different PCs, with different Docker desktop versions and on different networks

nghiant2710, 2022-01-19:
Hi @DrSchottky, thank you for the report. I think this is due to the temporary `shim` we install over `/bin/sh` to print image details before restoring the original `sh` binary. `apt-get` update calls `apt-key` which execs another process (via the function ExecGPGV) which is the one that needs `sh` to be a shell (`shim` is a script not shell).

We will fix this soon but in the meantime, you can work around this by starting the container with `sh` by doing `docker run --rm -it --entrypoint sh balenalib/raspberry-pi` (you can run bash later then).

DrSchottky, 2022-02-09:
Thank you, with `sh` it works fine

closed 2022-02-09 by DrSchottky