balena-io-library/base-images #748: Nvidia Jetson Nano base image doesn't support Jetpack 4.6

vipulgupta2048, 2022-02-04:
`jetson-nano-ubuntu`base image installs the [L4T package version 32.5](https://hub.docker.com/layers/balenalib/jetson-nano-ubuntu/latest/images/sha256-ce94a7c87bdff453384cb892453a8300f2f9d7513d3f50f1a30c9229240a3223?context=explore) released on January 2021. We need to upgrade this base image and install L4T package version 32.6.1 which supports Jetpack 4.6 as mentioned on Nvidia's L4T archive: https://developer.nvidia.com/embedded/linux-tegra-archive

mehalter, 2022-02-14:
@vipulgupta2048 @acostach has this issue been resolves with the latest update of the contracts?

nghiant2710, 2022-02-15:
Our base images are updated to Jetpack 4.6, closing.

closed 2022-02-15 by nghiant2710