balena-io-library/base-images #760: python base images not updated for months in dockerHub (systemic issue)

m10d, 2022-04-14:
I've searched the issues & forums for an issue covering this general issue - found several specific to a particular missing `tag` but nothing for what appears to be the systemic issue - hope this isn't a dup.

It appears to me that *numerous* python base images are not being properly built/pushed to `Docker Hub` (but the corresponding `node`, `dotnet` etc *are* published & fresh).

Two of many examples:

- `balenalib/amd64-alpine`  https://hub.docker.com/search?q=balenalib%2Famd64-alpine
    - `*-node`, `*-golang`, `*-dotnet` all updated within **days** ago (one day, at time of writing)
    - `balenalib/amd64-alpine-python` uniquely is _**3 months**_ ago
- `balenalib/amd64-ubuntu` https://hub.docker.com/search?q=balenalib%2Famd64-ubuntu
    - again **all** other languages are updated in the last few days (as of writing, `3 days` ago)
    - but `balenalib/amd64-ubuntu-python` was updated 4 **MONTHS** ago

Is this a known issue?  The docs clearly say that `python` support includes version `3.10.2` (https://www.balena.io/docs/reference/base-images/base-images/), but only the many-month old `3.10.0` is available as a published tag.

nghiant2710, 2022-04-14:
Hi @m10d, thank you for the report, since we just moved our base images to new build servers (still working on tuning the servers) so our base images are rebuilt less frequently and some tags might be missing due to build server issues, also languages have different release cycle so it affects how often a base image is updated.

For Python stack, we just added support for latest releases yesterday (https://github.com/balena-io/contracts/commit/53db535c3d0ec76cb2e1fe224d72e827046fb01c) and new base images will be released very soon, we will update the docs when new tags are available on Dockerhub (I will keep this issue opening until the new images are ready).

m10d, 2022-04-14:
Hi @nghiant2710 thank you for your quick reply.  

Your explanation makes sense with the exception of the size of missing time window.

Checking one random image, https://hub.docker.com/r/balenalib/amd64-ubuntu-golang/tags?page=23, I see rebuilt images going back somewhat consistently for the last several months.  IE there is no gap of > 1 month for `amd64-ubuntu-golang`...  Whereas Python stands (apparently) unique as having this large (several month) gap.

What piqued my interest to look into this was note _from_ several months ago in one of my base images; I `apt install`'d `python` manually in the `balenalib/amd64-ubuntu` (no `*-python`) based container, with a comment in my dockerfile that `"python ubuntu builds do not seem maintained(?)"`, comment checked in several months ago.   So it appears there were very infrequent builds comming out back then.

questions: 

1. what frequency do you expect these `*-python` builds to be being published now?  
1. Is there a reason `python` would be different cadence than other languages (even if the python `minor` version didn't change, the `ubuntu` system surely receives patches often that would be picked up on rebuild?
1. Does your explanation account for the fact that the other languages have not experienced this long gap in base image creation / publishing?

thank you.

jvelo, 2023-10-05:
Hello – Are there plans to publish 3.12 images ?
Thanks in advance