balena-io-library/base-images #762: manifest unknown

zoobot, 2022-04-15:
Hi,
I am getting this error when I try to run with this: 
`FROM balenalib/%%BALENA_MACHINE_NAME%%-python:3.8.13-focal-run`

`manifest for balenalib/intel-nuc-python:3.8.13-focal-run not found: manifest unknown: manifest unknown`

It looks like it should exist: [https://github.com/balena-io-library/base-images/blob/master/balena-base-images/python/intel-nuc/ubuntu/focal/3.8.13/run/Dockerfile](https://github.com/balena-io-library/base-images/blob/master/balena-base-images/python/intel-nuc/ubuntu/focal/3.8.13/run/Dockerfile)

Anything I need to change?

nghiant2710, 2022-04-16:
hi @zoobot,

We just added support for these python versions and the base images are being built so that tag is not available yet.
I will let you know when it is ready (probably in a day or two).

zoobot, 2022-04-18:
Thanks for the update @nghiant2710 ! 
I switched to python:3.8-stretch and that worked fine for my purposes.

closed 2022-05-04 by nghiant2710