balena-io-library/base-images #778: Python Alpine 3.15/3.16 images break packages with native C extensions

nojocodex, 2022-06-18:
It appears the Python Alpine 3.15 & 3.16 images include a Python `ctypes` linked against the wrong libffi.so. This breaks packages that use native C extensions, e.g. try `pip install pyyaml` on one of these images.

Alpine 3.15 updated libffi from 3.3 to 3.4, but AFAICT the balena base image `Dockerfile` still downloads python linked against 3.3. For example in `base-images/balena-base-images/python/amd64/alpine/3.16/3.10.5/build/Dockerfile`:

```curl -SLO "http://resin-packages.s3.amazonaws.com/python/v$PYTHON_VERSION/Python-$PYTHON_VERSION.linux-alpine-amd64-libffi3.3.tar.gz"```

Reproduction:

```
$ docker run -it balenalib/intel-nuc-alpine-python:3.10-3.16 /bin/bash
bash-5.1# python
Python 3.10.5 (main, Jun 11 2022, 15:30:19) [GCC 10.2.1 20201203] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import ctypes
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.10/ctypes/__init__.py", line 8, in <module>
    from _ctypes import Union, Structure, Array
ImportError: Error loading shared library libffi.so.7: No such file or directory (needed by /usr/local/lib/python3.10/lib-dynload/_ctypes.cpython-310-x86_64-linux-gnu.so)
>>>
bash-5.1# ls -al /usr/lib/libffi*
lrwxrwxrwx    1 root     root            15 Jun 17 03:10 /usr/lib/libffi.so.8 -> libffi.so.8.1.0
-rwxr-xr-x    1 root     root         47056 Oct 17  2021 /usr/lib/libffi.so.8.1.0
bash-5.1#
```

nghiant2710, 2022-06-19:
@nojocodex, thank you for the report. We are working on adding the new Python binary for latest Alpine linux release and will let you know when the new base images are available

nghiant2710, 2022-06-28:
@nojocodex, just want to let you know that, we have updated our Python base images. Please give it a try and let us know if you have any further question.

closed 2022-06-28 by nghiant2710