balena-io-library/base-images #785: Can't install packages in balenalib/raspberrypi3-debian

Joseph-Melberg, 2022-07-23:
I pulled the latest, my DockerFile has Run install_packages curl jq

When it runs, I get
`Step 2/7 : RUN install_packages curl jq
 ---> Running in c8dc2a0fee34
W: GPG error: http://deb.debian.org/debian bullseye InRelease: At least one invalid signature was encountered.
E: The repository 'http://deb.debian.org/debian bullseye InRelease' is not signed.
W: GPG error: http://deb.debian.org/debian-security bullseye-security InRelease: At least one invalid signature was encountered.
E: The repository 'http://deb.debian.org/debian-security bullseye-security InRelease' is not signed.
W: GPG error: http://deb.debian.org/debian bullseye-updates InRelease: At least one invalid signature was encountered.
E: The repository 'http://deb.debian.org/debian bullseye-updates InRelease' is not signed.
W: GPG error: http://archive.raspberrypi.org/debian bullseye InRelease: At least one invalid signature was encountered.
E: The repository 'http://archive.raspberrypi.org/debian bullseye InRelease' is not signed.
W: GPG error: http://archive.raspbian.org/raspbian bullseye InRelease: At least one invalid signature was encountered.
E: The repository 'http://archive.raspbian.org/raspbian bullseye InRelease' is not signed.
apt failed, retrying
W: GPG error: http://deb.debian.org/debian bullseye InRelease: At least one invalid signature was encountered.
E: The repository 'http://deb.debian.org/debian bullseye InRelease' is not signed.
W: GPG error: http://deb.debian.org/debian-security bullseye-security InRelease: At least one invalid signature was encountered.
E: The repository 'http://deb.debian.org/debian-security bullseye-security InRelease' is not signed.
W: GPG error: http://deb.debian.org/debian bullseye-updates InRelease: At least one invalid signature was encountered.
E: The repository 'http://deb.debian.org/debian bullseye-updates InRelease' is not signed.
W: GPG error: http://archive.raspbian.org/raspbian bullseye InRelease: At least one invalid signature was encountered.
E: The repository 'http://archive.raspbian.org/raspbian bullseye InRelease' is not signed.
W: GPG error: http://archive.raspberrypi.org/debian bullseye InRelease: At least one invalid signature was encountered.
E: The repository 'http://archive.raspberrypi.org/debian bullseye InRelease' is not signed.
apt failed, retrying
`

Am i doing something wrong?

nghiant2710, 2022-07-25:
hey @Joseph-Melberg, I couldn't reproduce this on my end or our servers, could be something wrong with the upstream repo at that time, can you please try again now?

Joseph-Melberg, 2022-07-25:
I pulled the latest balenalib/raspberrypi3-debian, ran the install_packages curl after  `docker run -it balenalib/raspberrypi3-debian sh` and I am still seeing the same issues.  

OS: Raspbian GNU/Linux 10 (buster) armv7l
Host: Raspberry Pi 3 Model B Plus Rev 1.3
Kernel: 5.10.17-v7+
Uptime: 2 days, 7 hours, 10 mins
Packages: 632 (dpkg)
Shell: bash 5.0.3
Terminal: /dev/pts/0
CPU: BCM2835 (4) @ 1.400GHz
Memory: 644MiB / 924MiB

Any idea what I am doing wrong?

nghiant2710, 2022-10-24:
Hey @pvbaudin, can you please share with us more details like what is the host OS? or what is the Docker version you are using and so on? Since I cannot reproduce your issue on my end 
```
nghiant2710@misc2:~$ docker run -ti --rm balenalib/rpi-raspbian:latest bash
Unable to find image 'balenalib/rpi-raspbian:latest' locally
latest: Pulling from balenalib/rpi-raspbian
5b5a2e86ef84: Pull complete
30869e65a46e: Pull complete
0f6d901b7bb0: Pull complete
9f443281bb79: Pull complete
015d6629e076: Pull complete
9e1d102411fe: Pull complete
605b78904a87: Pull complete
a6f5c5588b6c: Pull complete
Digest: sha256:cfcde2b226297985ae45d438e206a192885a2e7d73a98f0da809ab60b7cb879b
Status: Downloaded newer image for balenalib/rpi-raspbian:latest
WARNING: The requested image's platform (linux/arm/v6) does not match the detected host platform (linux/amd64) and no specific platform was requested
root@3ce2414f7762:/# apt-get update
Get:1 http://archive.raspbian.org/raspbian bullseye InRelease [15.0 kB]
Get:2 http://archive.raspberrypi.org/debian bullseye InRelease [23.6 kB]
Get:3 http://archive.raspbian.org/raspbian bullseye/firmware armhf Packages [1201 B]
Get:4 http://archive.raspbian.org/raspbian bullseye/rpi armhf Packages [1299 B]
Get:5 http://archive.raspbian.org/raspbian bullseye/non-free armhf Packages [130 kB]
Get:6 http://archive.raspbian.org/raspbian bullseye/main armhf Packages [18.3 MB]
Get:7 http://archive.raspbian.org/raspbian bullseye/contrib armhf Packages [70.3 kB]
Get:8 http://archive.raspberrypi.org/debian bullseye/main armhf Packages [307 kB]
Fetched 18.8 MB in 8s (2470 kB/s)
Reading package lists... Done
W: Skipping acquire of configured file 'ui/binary-armhf/Packages' as repository 'http://archive.raspberrypi.org/debian bullseye InRelease' doesn't have the component 'ui' (component misspelt in sources.list?)
root@3ce2414f7762:/#
```

pvbaudin, 2022-10-24:
@nghiant2710 

> Docker version 20.10.20, build 9fdeb9c
Host OS:
> Distributor ID:	Raspbian
> Description:	Raspbian GNU/Linux 10 (buster)
> Release:	10
> Codename:	buster

Hardware: Raspberry Pi 4

Tried using install_packages in dockerfile, also tried apt-get update in dockerfile and in interactive shell. Tried manually downloading GPG key. nothing worked.

jumarti, 2023-04-27:
This worked for me, from https://copyprogramming.com/howto/atleast-one-invalid-signature-was-encountered

For Raspbian, upgrade  libseccomp  manually on the host system by using:
```
curl http://ftp.us.debian.org/debian/pool/main/libs/libseccomp/libseccomp2_2.5.1-1_armhf.deb --output libseccomp2_2.5.1-1_armhf.deb
sudo dpkg -i libseccomp2_2.5.1-1_armhf.deb
```