balena-io-library/base-images #794: Any images for BeagleBone Black + Ubuntu 14.04.3 LTS (Trusty Tahr)?

jamesbraza, 2022-09-28:
I have a BeagleBone Black with Ubuntu 14.04.3 LTS (Trusty Tahr).

I see [here](https://hub.docker.com/r/balenalib/beaglebone-black-ubuntu) that Tahr is not currently on the list.  Might there be a Tahr somewhere, and the list is out of date?

And if there isn't a Tahr, can you possibly point me in a direction where I can find a Tahr image?

jamesbraza, 2022-09-28:
Also, from [this Docker Hub filter](https://hub.docker.com/r/balenalib/beaglebone-black-ubuntu-python/tags?page=2&ordering=-name&name=xenial-run-202), I can't find `3.10.5-xenial-run-20220923` anywhere, despite it being [here](https://github.com/balena-io-library/base-images/tree/master/balena-base-images/python/beaglebone-black/ubuntu/xenial).

Can you please push the Xenial 3.10 image to Docker Hub? 🤓 Thank you in advance

klutchell, 2023-09-12:
Attempting a re-run of the BB python jobs here: https://github.com/balena-io-library/base-images/actions/runs/6162515629

klutchell, 2023-09-12:
See this [comment](https://github.com/balena-io-library/base-images/issues/904#issuecomment-1716173651) regarding missing python variants.

closed 2023-09-12 by klutchell