balena-io-library/base-images #877: Rpi4 bullseye image is missing raspbian source

Geethree, 2023-08-08:
# What
A new `balenalib/raspberrypi4-64-debian:bullseye` image came out a [few days ago](https://hub.docker.com/r/balenalib/raspberrypi4-64-debian/tags?page=1&name=bullseye).. this is missing the raspbian sources. I don't see any changes in the underlying dockerfile where the raspi source is added

https://github.com/balena-io-library/base-images/blob/1bcdc7c3a216f6e298343437def9414ad100a0da/balena-base-images/device-base/raspberrypi4-64/debian/bullseye/build/Dockerfile#L3-L4

Though as you can see below, this is not the case.

# Differences

```
root@fc8fa89:~# balena pull balenalib/raspberrypi4-64-debian:bullseye-20230530
bullseye-20230530: Pulling from balenalib/raspberrypi4-64-debian
Digest: sha256:38c05deda9d2d99ecb07ec195762872def57cd6f2828c09caa691b3a601df6d5
Status: Image is up to date for balenalib/raspberrypi4-64-debian:bullseye-20230530
docker.io/balenalib/raspberrypi4-64-debian:bullseye-20230530
root@fc8fa89:~# balena run --rm -it --entrypoint /bin/ls balenalib/raspberrypi4-64-debian@sha256:38c05deda9d2d99ecb07ec195762872def57cd6f2828c09caa691b3a601df6d5 /etc/apt/sources.list.d
raspi.list
root@fc8fa89:~# balena pull balenalib/raspberrypi4-64-debian:bullseye         
bullseye: Pulling from balenalib/raspberrypi4-64-debian
Digest: sha256:a087d20c837ea39abdededa61cb69ed960685e381385b1da2404f687a161345e
Status: Image is up to date for balenalib/raspberrypi4-64-debian:bullseye
docker.io/balenalib/raspberrypi4-64-debian:bullseye
root@fc8fa89:~# balena run --rm -it --entrypoint /bin/ls balenalib/raspberrypi4-64-debian@sha256:a087d20c837ea39abdededa61cb69ed960685e381385b1da2404f687a161345e /etc/apt/sources.list.d
```

klutchell, 2023-08-21:
Hey @Geethree , I confirmed that I am seeing the same issue and will be looking into it!

klutchell, 2023-08-22:
This has been resolved, thank you for bringing it to our attention!

closed 2023-08-22 by klutchell