balena-io-library/base-images #904: error when building image with: /balena-base-images/python/i386/ubuntu/bionic/3.10.10/build/Dockerfile

CookingANut, 2023-08-31:
meet issue with executing python command:
32.55 + python3 get-pip.py --disable-pip-version-check --no-cache-dir --no-compile pip==23.0 setuptools==67.2.0
32.55 python3: /lib/i386-linux-gnu/libc.so.6: version `GLIBC_2.28' not found (required by /usr/local/lib/libpython3.10.so.1.0)

CookingANut, 2023-08-31:
I tried "apt-get install -y libc6 libc6-dev", no help

klutchell, 2023-09-12:
It seems that the older ubuntu variants (bionic and xenial) are using glibc versions too old for our [python binaries](https://github.com/product-os/python-build).

We will either have to build new binaries that are compatible with glibc 2.24 or limit which ubuntu variants receive python.