balena-io-library/base-images #954: Support for Node v20 image

fritzdimmel, 2023-10-11:
Hi, as Node16 just went to EOL, we updated our application to Node20. Right now the latest balenalib/%%BALENA_MACHINE_NAME%%-node:19-build image is v19.
Is there already a v20 in preparation so that we can move our devices to the v20? Any release date?

I see that v20 is going to be LTS by Oct 24 accourding to [Releases](https://nodejs.dev/en/about/releases/) but it would be good to be able to start now instead of going to v18 and in a month to v20.
Thanks
Fritz

Iznogohul, 2023-10-14:
@fritzdimmel 

Completely agree with your point. It would be highly beneficial for Balena to offer a continuous range of Node.js images, including the latest LTS version (currently 18.18.2) and keeping pace with the latest edge releases (like 20.8.1). This approach would help ensure that we are not missing out on important security patches and updates.

For instance, the recent CVE-2023-44487, which led to a significant DDoS attack, underscores the urgency of staying up-to-date with Node.js versions. Having access to these updates in a timely manner is crucial for maintaining the security and stability of our applications.

fritzdimmel, 2024-03-11:
Any news on that from the balena team? It's 6 month now since the release of Node 20 as LTS...
Thanks

Petersdavis, 2024-04-04:
I think that there is a path to accomplish this now.  I am able to get node20 working in bionic  by following these build instructions:
https://stackoverflow.com/a/77247394/6425571

It should be fairly trivial to port the solution to a base image.

Iznogohul, 2024-04-17:
@fritzdimmel @Petersdavis 
Balena supports node 20.12.0  for roughly ~ 1 week now https://github.com/balena-io/contracts/pull/419
Was able to pull balenalib/aarch64-debian-node:20

hanyufoodles, 2024-09-26:
Is there plan to support node 20 for other builds? For example alpine?

Iznogohul, 2024-12-04:
@hanyufoodles Based on https://github.com/balena-io/contracts/pull/419 there is plans for alpine builds once the jenkins job can work or that was the case when this pull opened.
I believe it is time to support node v22 as well both for alpine and debian base image