balena-io-library/base-images #1007: Node LTS Image Timeline

deanMike, 2024-01-22:
I know there is already a similar issue here: #954

It didn't look like there was anyone from Balena involved in that conversation though and since then Node 20 has become the Active LTS. I'm trying to see when we might expect base-images for the current active LTS release of Node.js to become available?

nerdbaggy, 2024-01-30:
Looking for the same info, management would like us to be on the LTS version of node

Petersdavis, 2024-04-04:
we got node20 running on bionic by following this thread:
https://stackoverflow.com/a/77247394/6425571

Iznogohul, 2024-04-17:
@deanMike  @nerdbaggy 
Balena supports node 20.12.0  for roughly ~ 1 week now https://github.com/balena-io/contracts/pull/419
Was able to pull balenalib/aarch64-debian-node:20

deanMike, 2024-04-17:
Thanks so much!

closed 2024-04-17 by deanMike